morrownr/8821cu-20210118 #79: help with kali wireless

blkops99, 2022-12-12:
i am VERY new to linux, i cannot seem to enablke my realtek 8821CE device  i have tried to install the compat driver among other "things"  i really need some help  i am connected through virtual box using my laptops wireless so i have connectivity but i wish to use some aircrack and i cannot see any wireless networks  help!

morrownr, 2022-12-12:
Hi @blkops99 

Welcome to Linux.

If you have installed the driver in this repo, please run the following from a command line interest:

$ sudo ./remove-driver.sh

Once that is complete, delete the driver directory.

The reason for the above. This is not the driver for the 8821ce chipset. With that said, let me try to help.

The driver for the 8821ce chipset is an in-kernel driver that falls under rtw88. You card should be active if you are running a recent version of Kali. You can determine if it is active by running:

$ iw dev

Try all of the above and tell me what happens.

Nick

blkops99, 2022-12-13:
this is maddening - however i do love the flexibility, the tools, and the privacy Kali offers - i REALLY appreciate the help!

ok so here is what i did and what happened

┌──(blkops㉿DESKTOP-SO13MHR)-[~]└─$ sudo ./remove-driver.sh  
[sudo] password for blkops:
sudo: ./remove-driver.sh: command not found

┌──(blkops㉿DESKTOP-SO13MHR)-[~]
└─$ su root                
Password:
┌──(root㉿DESKTOP-SO13MHR)-[/home/blkops]
└─# sudo ./remove-driver.sh
sudo: ./remove-driver.sh: command not found

┌──(root㉿DESKTOP-SO13MHR)-[/home/blkops]
└─#
i also have a TpLink USB AC600 Mini wireless adaptor ver3 that i cant get to work either!
i am running Kali in a virtual box

Sent with Proton Mail secure email.

morrownr, 2022-12-13:
Opps...

My bad. You told me you were a newbie and I failed to provide full instructions.

> sudo: ./remove-driver.sh: command not found

Ctrl-Alt-t
$ cd ~/src/8821cu-20210118
$ sudo ./remove-driver.sh

Step 1 above opens a command line interface for you to enter the next two lines one at a time followed by Enter. (don't enter the prompt $)
Step 2 moves you into the directory where you downloaded the driver if you followed the installation steps.
Step 3 runs the script to full clean out everything that was installed. Gotta keep the system clean. You can then close the command line interface `$ exit` and then use the file manager to delete the driver folder/directory as the driver will not be needed.

> i am running Kali in a virtual box

I have a link in the FAQ here that talks about Kali and VirtualBox. It might have a hint or two.

Now that the above is taken care of and you have rebooted, let's see if you have a wifi interface based on the in-kernel driver:

Ctrl-Alt-t
$ iw dev

If you have a wifi interface, you will see something like the below:

```
	Interface wlan0
		ifindex 4
		wdev 0x1
		addr ee:44:06:89:29:2f
		type managed
		txpower -100.00 dBm

```

blkops99, 2022-12-14:
Ok. Here is what I did. - I REMOVED KALI AND ALL FILES and reinstalled. I have a fresh install and the same problem- only this time I did not do anything. So, where should I start? I have connectivity- just no wireless. Thank you

Sent with Proton Mail secure email.

blkops99, 2023-01-13:
how are ya! so  here is what i have - i did a full kali install from iso the installer image into VMWare Workstation Pro 16.2.5 build-20904516, went very smooth - all up until the wireless part  i have a TpLink T2U and a Linksys WUSB6300 V2. i have installed your driver and removed the old i have done what i believe to be every combination available to me in the VM network manager and the machine settings  i have updated upgraded etc - im stuck  i tried to add a usb device to vm and it said no that there werent any spots available - so i removed it and reistalled to no avail  please HELP

Sent with Proton Mail secure email.

blkops99, 2023-01-13:
i also can get it to connect to my wifi card - but wont see it as wifi  both adaptors show up in lsusb but iwconfig say no wireless extensions

Sent with Proton Mail secure email.

morrownr, 2023-01-14:
> i also can get it to connect to my wifi card - but wont see it as wifi  both adaptors show up in lsusb but iwconfig say no wireless extensions

iwconfig is depreciated and no longer maintained. What does the following say:

$ iw dev

Have you considered going with a dual boot setup instead of using a VM?

blkops99, 2023-01-14:
iw dev says nothing - i tried a dual boot - kali was like an irascible teenager! it was always being difficult - however, as im asure you can tell i am VERY new to linux - pretty good with windows and mac but linux.... and i LOVE it! i did change fro  the iso to the prepackaged kali for vmware - still can get my adaptor installed  thanks so much for your help
what next?

i tried a dual boot  -  kali was like a

Sent with Proton Mail secure email.

morrownr, 2023-01-14:
> iw dev says nothing 

That just means that no wifi interface was created which means the driver is not seeing the adapter.

> i tried a dual boot - kali was like an irascible teenager!

Yeah, Kali can be a challenge. Not a good newbie distro. You can always come back to it after learning on an easier distro. The following might be good place to start:

https://linuxmint.com/

Make sure you study up on how to do dual boot. Also, keep in mind that you can install the apps you are looking for in Kali on almost any distro.

Sorry about not being much help with VMWare but I have never used it.

blkops99, 2023-01-14:
I really appreciate your help! i also have a tailsinstall on usb it is SWEET! I am taking classes and go to every tutorial i can! I really like it

Sent with Proton Mail secure email.

morrownr, 2023-01-14:
It can be a lot to learn and learning it best done over a period of time so as to let things soak in. Stay at it, you will get there.

blkops99, 2023-02-19:
so, i took your advice and got myself a nice Debian 10 VPS and an installation of RedHat  I am doing online tutorials and learning as quickly as i can 

but Kali FASCINATES me - I LOVE IT and i tinker with it quite often - I WILL learn it! I now have it installed on a Dell Inspirion - all by itself!

I CANNOT get Kali to recognize my adapters - it will only recognize my internal card. I have been at this for quite some time! Could you help me?
 I have included some info in the folder
i did not install those dkms modules
the usb0 is a tethered android so i can use the internal card for monitor mode

let me know and i really appreciate it

Sent with Proton Mail secure email.

─$ iw dev  
phy#0
	Interface wlan0
		ifindex 2
		wdev 0x1
		addr 3c:55:76:a3:6d:25
		type monitor
		channel 11 (2462 MHz), width: 20 MHz (no HT), center1: 2462 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	00
			
			ip a:   
			
			1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: wlan0: <BROADCAST,ALLMULTI,PROMISC,NOTRAILERS,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ieee802.11/radiotap 3c:55:76:a3:6d:25 brd ff:ff:ff:ff:ff:ff
3: usb0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    link/ether 8e:e6:c1:c5:24:fe brd ff:ff:ff:ff:ff:ff
    inet 192.168.205.28/24 brd 192.168.205.255 scope global dynamic noprefixroute usb0
       valid_lft 2206sec preferred_lft 2206sec
    inet6 fe80::961a:1b93:fd6d:234b/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
       
       dkms status
lime-forensics/1.9.1-5, 6.0.0-kali3-amd64, x86_64: installed
rtl8821cu/5.12.0.4, 6.0.0-kali6-amd64, x86_64: installed
xtrx/0.0.1+git20190320.5ae3a3e-3.2, 6.0.0-kali3-amd64, x86_64: installed
xtrx/0.0.1+git20190320.5ae3a3e-3.2, 6.0.0-kali6-amd64, x86_64: installed

lsusb:

Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 0c45:6730 Microdia Integrated_Webcam_HD
Bus 003 Device 002: ID 13b1:0045 Linksys WUSB6300 V2
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 2357:0115 TP-Link Archer T4U ver.3
Bus 001 Device 009: ID 22b8:2e24 Motorola PCS moto g pure
Bus 001 Device 006: ID 2357:011f TP-Link 802.11ac WLAN Adapter 
Bus 001 Device 005: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 004: ID 0bda:c829 Realtek Semiconductor Corp. Bluetooth Radio 
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub

lsmod
Module                  Size  Used by
rndis_host             24576  0
cdc_ether              24576  1 rndis_host
usbnet                 57344  2 rndis_host,cdc_ether
mii                    16384  1 usbnet
[... 158 lines cut ...]
ANSI_COLOR="1;31"

 hostnamectl
 Static hostname: kali
       Icon name: computer-laptop
         Chassis: laptop 💻
      Machine ID: fd0c50a14f6d4335acc1d0989e243aa2
         Boot ID: 8915c838a98b4b168fb81c393a2090d4
Operating System: Kali GNU/Linux Rolling          
          Kernel: Linux 6.0.0-kali6-amd64
    Architecture: x86-64
 Hardware Vendor: Dell Inc.
  Hardware Model: Inspiron 15 3515
Firmware Version: 1.8.0

cat /proc/version
Linux version 6.0.0-kali6-amd64 ***@***.***) (gcc-12 (Debian 12.2.0-10) 12.2.0, GNU ld (GNU Binutils for Debian) 2.39.50.20221208) #1 SMP PREEMPT_DYNAMIC Debian 6.0.12-1kali1 (2022-12-19)

No LSB modules are available.
Distributor ID:	Kali
Description:	Kali GNU/Linux Rolling
Release:	2022.4
Codename:	kali-rolling

lsb_release -a

morrownr, 2023-02-19:
> I CANNOT get Kali to recognize my adapters - it will only recognize my internal card. I have been at this for quite some time! Could you help me?

This smells like a Secure Boot issue. Is secure boot on in the bios?

Run and post the results of the following:

$ mokutil --sb-state

blkops99, 2023-02-19:
SecureBoot disabled
Platform is in Setup Mode

i also removed all those modules in dkms ive never seen before - except yours! lol

Sent with Proton Mail secure email.

morrownr, 2023-02-19:
Hi again @blkops99 

Well, it is hard to provide tech support this way but...

What I need to do is back up and reexamine the situation. I just skimmed the thread again.

For my sake, let's try to keep this simple so I can get a handle on it. Tell me if the following is correct:

You have a Dell laptop that only has Kali installed thus preventing the need for a VM?

On this system, you have an internal wifi card and an extrenal usb wifi adapter?

On the above Dell system with Kali on it, run and post the results of the following:

$ lspci
$ lsusb
$ iw dev
$ dkms status
# uname -r

That will help me get a handle on the situation, I hope.

blkops99, 2023-02-19:
thank you so very much

Dell laptop with Kali only
internal wifi
3 external adapters (i dont care which one two or three you get going!)
Archer TPlink T2U v3
Archer TPlink AC1300 MU-MIMO T4U
Linksys WUSB6300 V2

attached are all runs

again - thank you

i had the linksys and the T2U both working in a vm  i cant get anything to work now

Sent with Proton Mail secure email.

└─$ lspci   
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Root Complex
00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 IOMMU
00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge
00:01.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 PCIe GPP Bridge [6:0]
00:01.7 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 PCIe GPP Bridge [6:0]
00:08.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge
00:08.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Internal PCIe GPP Bridge 0 to Bus A
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller (rev 61)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge (rev 51)
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 0
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 1
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 2
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 3
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 4
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 5
00:18.6 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 6
00:18.7 Host bridge: Advanced Micro Devices, Inc. [AMD] Raven/Raven2 Device 24: Function 7
01:00.0 Non-Volatile memory controller: KIOXIA Corporation Device 000c
02:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8821CE 802.11ac PCIe Wireless Network Adapter
03:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Picasso/Raven 2 [Radeon Vega Series / Radeon Vega Mobile Series] (rev c2)
03:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Raven/Raven2/Fenghuang HDMI/DP Audio Controller
03:00.2 Encryption controller: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) Platform Security Processor
03:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Raven USB 3.1
03:00.4 USB controller: Advanced Micro Devices, Inc. [AMD] Raven USB 3.1
03:00.5 Multimedia controller: Advanced Micro Devices, Inc. [AMD] ACP/ACP3X/ACP6x Audio Coprocessor
03:00.6 Audio device: Advanced Micro Devices, Inc. [AMD] Family 17h/19h HD Audio Controller

Bus 001 Device 007: ID 22b8:2e82 Motorola PCS XT1541 [Moto G 3rd Gen]

		type managed
		channel 7 (2442 MHz), width: 20 MHz, center1: 2442 MHz

└─$ dkms status
rtl8812au/5.13.6, 6.0.0-kali6-amd64, x86_64: installed

└─$ uname -r                 
6.0.0-kali6-amd64