morrownr/8821cu-20210118 #80: failed on centos stream 9 (kernerl: 5.14)

sinscut, 2023-01-01:
error while compling:
implicit declaration of function 'conplete_and_exit'
cc1: some warnings being treated as errors.

morrownr, 2023-01-01:
Hi @sinscut 

Is your centos based on RHEL?

sinscut, 2023-01-02:
Thank you for your reply @morrownr 
The official [website](https://www.centos.org/centos-stream/) says it is a midstream between fedora and RHEL.

morrownr, 2023-01-02:
Well, that does not sound good. The problem with RHEL and any distro that does the same bad stuff that it does with backporting is that it would take far more time to support than I have available. In fact, unless you have a good reason to use RHEL or Centos, I recommend you use something more mainstream like Fedora, Ubuntu, Debian, Manjaro, or many more.

You might have luck with adapters that use in-kernel drivers:

https://github.com/morrownr/USB-WiFi

Reading menu items 1 and 2 would be a good start.

Good luck,

Nick