morrownr/8821cu-20210118 #81: wireless driver for kernel 6.0

Hell0Mini, 2023-01-02:


Hell0Mini, 2023-01-02:
I have D-link DWA 171 with chipset rtl8821CU_rtl8731AU with kali 6.0 please update the driver for 6.0 kernel. or help me how can i do.

morrownr, 2023-01-02:
Hello @Hell0Mini 

I'll be glad to help.

Are you using a computer with an x86_64 or arm64 processor... or other?

Have you checked the device ID of your adapter to make sure this is the correct driver?

paralin, 2023-01-03:
FYI it works with 6.1.2 :)

morrownr, 2023-01-03:
@paralin 

Is this something Kali messed up?

paralin, 2023-01-03:
This particular issue needs more information - like an error message at least.

alx52, 2023-01-04:
Fedora 37, kernel 6.0.15 works good, hotspot works too. Thank you all😀!

checor, 2023-01-15:
EndeavourOS / Arch with kernel 6.1.6 works. I lost connection after a kernel upgrade, but a clean reinstall of the driver and a reboot made the wifi back. 

Keep up the good work!

morrownr, 2023-01-15:
Hi @checor 

> I lost connection after a kernel upgrade

Do you remember the specifics of the situation? The reason I ask is because we are working on a new version of this driver right now in a private repo. Making the support for kernel upgrades more robust is one thing we are working on.

FYI: I think the new driver is impressive. Even MU-MIMO is working. You can watch wavemon while you start a download or iperf3 test and the signal strength/line quality gets better as the router/AP determines where it needs to concentrate signal. I think it is cool.

Do you want to test?

Nick