morrownr/8821cu-20210118 #82: Struggle to build on Pi 4 B (1GB)

colincdean, 2023-01-02:
When running install-driver on  Raspberry Pi 4 model B with 1GB RAM running 64-bit Raspberry Pi OS 11, compilation is incredibly slow and may fail (with a SIGKILL).  This is reproducible.  When used previously with same hardware running 32-bit Raspberry Pi OS 10, the problem didn't occur.

Appears to be because the "-j5" make flag causes all available RAM to be used and the device to start swapping.   A simple workaround is to edit `dkms.conf`, replacing `-j$(nproc)` by `-j1` (which actually makes it faster and complete successfully).

Is there some way the install script could automatically reduce the number of simultaneous jobs in this scenario?

P.S. Thanks for supporting these adapters.  I have 8821cu (as client with BrosTrend AC650 AC5L) and 88x2bu (as AP with BrosTrend AC1200 AC3L) working well on the same Pi 4, used as firewall/gateway.

morrownr, 2023-01-02:
Hi @colincdean 

I was unaware this was happening on the 1gb Pi's. I actually has a 1 gb Pi3B so I need to test this. My Pi4B has 4 gb of ram so it won't help.

After reading your message, I thought "it sure would be good if Colin was testing the new version of this driver." Right now myself and about a dozen others are testing a new version of this driver in a non-publc repo with an expect release date of later this month. Would you like access? There are some pretty smart guys over there. If you repost this issue over these, we might be able to come up with a solution and the driver needs testing anyway.

The new version is probably the most feature complete, stable driver I've ever seen for Linux. It even does MU-MIMO and it does it well. As much as I like the in-kernel drivers, for now this new driver appears to be the single best usb wifi driver there is.

Let me know if I can add you to the repo for the new driver and we will come up with a solution.

Nick

colincdean, 2023-01-02:
Hi @morrownr 

Yes, I'd be happy to test the new driver.  I actually have the Pi 4 in question in a cardboard box waiting for me to take it back to its normal location 30 miles away, which can wait a few days, so this is good timing for me.

Colin

morrownr, 2023-01-02:
Colin,

You should get an email shortly that will guide you to the site. To access the site you will have to log into github so that it knows it is you.

Once you are in the repo, post your issue and we will see what a good solution is.

Nick

SubZero5, 2023-01-13:
Hi @morrownr

I would also be happy to test the new driver. 
May I also request an the invite?

Thanks