morrownr/8821cu-20210118 #83: Lag problem when watching video

murattahtacii, 2023-01-05:
I am a Fedora 37 user and I don't know much about linux. I have rtl8811cu. When I install this drive, there is an instant lag of about 2 seconds when listening to music or watching videos. I think the reason is because the instant cpu consumption increases. It does not happen when I connect with my adapter that only supports 2.4 ghz. I have been trying to solve it for days but to no avail. Another problem is that although I set the country setting, 5ghz does not appear in the wifi list. It says numbers like wlp0s20f0u2 in parentheses. Thanks in advance.

morrownr, 2023-01-06:
Hi @murattahtacii 

This is interesting. Would you be interested in trying a new version of this driver? It is currently in testing and is not public yet. You would have to log into github to access the driver but that might be a good way to sort this out.

murattahtacii, 2023-01-08:
That would be nice. I'll try it when I can. Thank you, good work.

morrownr, 2023-01-08:
I have added you to the private repo. You should get an email with instructions. Repost the problem there.

murattahtacii, 2023-02-02:
Good afternoon, I was able to solve my problem above by turning off the wifi card for Dual NIC ones. However, I could not find how to uninstall the RTL8111/8168/8411 driver. I have to turn it off every time I restart the computer. I am currently using opensuse.

murattahtacii, 2023-02-02:
Sure. sudo /usr/sbin/ip link set wlp2s0 down
I turn it off with this code but it restarts every time I start the computer.

`$ lsmod`

Module                  Size  Used by
binfmt_misc            28672  1
rfcomm                 90112  4
snd_seq_dummy          16384  0
snd_hrtimer            16384  1
snd_seq                98304  7 snd_seq_dummy
snd_seq_device         16384  1 snd_seq
af_packet              61440  14
nft_fib_inet           16384  1
nft_fib_ipv4           16384  1 nft_fib_inet
nft_fib_ipv6           16384  1 nft_fib_inet
nft_fib                16384  3 nft_fib_ipv6,nft_fib_ipv4,nft_fib_inet
nft_reject_inet        16384  6
nf_reject_ipv4         16384  1 nft_reject_inet
nf_reject_ipv6         20480  1 nft_reject_inet
nft_reject             16384  1 nft_reject_inet
nft_ct                 24576  7
nft_chain_nat          16384  3
cmac                   16384  3
algif_hash             16384  1
algif_skcipher         16384  1
af_alg                 36864  6 algif_hash,algif_skcipher
nf_tables             286720  236 nft_ct,nft_reject_inet,nft_fib_ipv6,nft_fib_ipv4,nft_chain_nat,nft_reject,nft_fib,nft_fib_inet
ip6table_nat           16384  0
ip6table_mangle        16384  0
ip6table_raw           16384  0
ip6table_security      16384  0
iptable_nat            16384  0
nf_nat                 57344  3 ip6table_nat,nft_chain_nat,iptable_nat
nf_conntrack          188416  2 nf_nat,nft_ct
bnep                   28672  2
nf_defrag_ipv6         24576  1 nf_conntrack
nf_defrag_ipv4         16384  1 nf_conntrack
iptable_mangle         16384  0
iptable_raw            16384  0
iptable_security       16384  0
hid_generic            16384  0
usbhid                 73728  0
nfnetlink              20480  2 nf_tables
ip6table_filter        16384  0
ip6_tables             36864  5 ip6table_filter,ip6table_raw,ip6table_nat,ip6table_mangle,ip6table_security
iptable_filter         16384  0
bpfilter               16384  0
btusb                  65536  0
btrtl                  28672  1 btusb
btbcm                  24576  1 btusb
btintel                45056  1 btusb
btmtk                  16384  1 btusb
bluetooth             925696  36 btrtl,btmtk,btintel,btbcm,bnep,btusb,rfcomm
uvcvideo              163840  0
videobuf2_vmalloc      20480  1 uvcvideo
videobuf2_memops       20480  1 videobuf2_vmalloc
videobuf2_v4l2         40960  1 uvcvideo
rtsx_usb_sdmmc         32768  0
videobuf2_common       90112  4 videobuf2_vmalloc,videobuf2_v4l2,uvcvideo,videobuf2_memops
videodev              323584  3 videobuf2_v4l2,uvcvideo,videobuf2_common
mmc_core              233472  1 rtsx_usb_sdmmc
rtsx_usb_ms            24576  0
ecdh_generic           16384  2 bluetooth
memstick               24576  1 rtsx_usb_ms
mc                     81920  4 videodev,videobuf2_v4l2,uvcvideo,videobuf2_common
rtsx_usb               32768  2 rtsx_usb_sdmmc,rtsx_usb_ms
8821cu               3727360  0
qrtr                   53248  4
snd_sof_pci_intel_skl    16384  0
[... 115 lines cut ...]
crypto_simd            16384  1 aesni_intel
cryptd                 28672  3 crypto_simd,ghash_clmulni_intel
cdrom                  77824  1 sr_mod
rc_core                69632  1 cec
usbcore               380928  7 xhci_hcd,usbhid,8821cu,rtsx_usb,uvcvideo,btusb,xhci_pci
ttm                    94208  3 drm_ttm_helper,i915,nouveau
video                  69632  2 i915,nouveau
wmi                    45056  5 video,intel_wmi_thunderbolt,wmi_bmof,mxm_wmi,nouveau
battery                28672  0
button                 24576  1 nouveau
serio_raw              20480  0
btrfs                1966080  1
blake2b_generic        20480  0
libcrc32c              16384  5 nf_conntrack,nf_nat,btrfs,nf_tables,xfs
crc32c_intel           24576  2
xor                    24576  1 btrfs
raid6_pq              122880  1 btrfs
sg                     49152  0
dm_multipath           45056  0
dm_mod                196608  1 dm_multipath
scsi_dh_rdac           16384  0
scsi_dh_emc            16384  0
scsi_dh_alua           20480  0
msr                    16384  0
efivarfs               24576  1

murattahtacii, 2023-02-02:
I solved the problem by turning it off in bios. If you have a different solution, it would be good if you write it down. Thanks for your help.

morrownr, 2023-02-02:
> I was going to suggest looking to see if you could turn the internal wifi or ethernet, whichever one is the problem, in the bios as that is the cleanest solution.

The 3 drivers I saw were:

8821cu 3727360 0
rtl8723be 126976 0
r8169 98304 0

I suppose it was the rtl8723be that you wanted to blacklist. An easy way to do it is to use the 8821cu.conf file that can be opened and edited with:

$ sudo ./edit-options.sh

You should see a line that starts with `blacklist`. Under that line, add a line like:

blacklist rtl8723be

...and save/reboot.

Remember to upgrade to the new drivers.

Cheers

murattahtacii, 2023-02-02:
Thank you, I have no more problems.