morrownr/8821cu-20210118 #86: Could not set channel for kernel driver - unable to run hostapd in ac mode on PI 4

tkrah, 2023-01-14:
Hi,

got a PI 400 and an usb stick with that usb id 0bda:c811 which should be supported by this driver (it works in client mode and I can get non ac mode working - but VHT is struggeling):
Driver was loaded with that options:

`options 8821cu rtw_drv_log_level=1 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=0 rtw_dfs_region_domain=3`

iw list has this:

```
Wiphy phy1
	wiphy index: 1
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CMAC (00-0f-ac:6)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CCMP-256 (00-0f-ac:10)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
		* CMAC-256 (00-0f-ac:13)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1862
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 150 Mbps
		HT TX/RX MCS rate indexes supported: 0-7
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (16.0 dBm)
			* 2417 MHz [2] (16.0 dBm)
			* 2422 MHz [3] (16.0 dBm)
			* 2427 MHz [4] (16.0 dBm)
[... 248 lines cut ...]
```

iw list has channel 36 and the freq supported, but the driver fails to set the channel:

Could not set channel for kernel driver

Ideas?

PS: running on a pi 400:

uname: Linux pi400 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
dkms: rtl8821cu, 5.12.0, 5.15.84-v8+, aarch64: installed
iw dev:

```
phy#1
	Interface wlx60fb006444a5
		ifindex 4
		wdev 0x100000001
		addr 60:fb:00:64:44:a5
		type managed
		txpower -100.00 dBm
```

latest master branch.

morrownr, 2023-01-14:
Hi @tkrah 

I'm a little busy today but maybe we can make progress anyway. Can I get you to go to the site Main Menu:

https://github.com/morrownr/USB-WiFi

And select menu item 9. It should say...

How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters

That document has a vast amount of info about setting up AP mode on a Pi. It has example files for hostapd.conf for both 5 GHz and 2.4 GHz and has specific about your adapter and many more. Read that and get back to me.

Nick

morrownr, 2023-01-14:
Made a mistake above...

How_to_Build_a_Bridged_Wireless_Access_Point