morrownr/8821cu-20210118 #87: Gembird WNP-UA-008 (8811cu) not recognized in 6.2-rc4 kernel

wowa-2017, 2023-01-20:
Trying to get working subj adapter on Ubuntu. Starting with 22.10, I upgraded kernel to current 6.2-rc4 version, lsusb shows
`Bus 001 Device 007: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC`
but iw dev does not show the device. 
dmesg has
`[  555.262426] usb 1-12: new high-speed USB device number 7 using xhci_hcd
[  555.410841] usb 1-12: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
[  555.410856] usb 1-12: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  555.410863] usb 1-12: Product: 802.11ac NIC
[  555.410869] usb 1-12: Manufacturer: Realtek
[  555.410874] usb 1-12: SerialNumber: 123456`
for it.
I thought those are regular vid/pid for the 881cu, what should I check or try to get it working?

wowa-2017, 2023-01-20:
Upd: installed this driver on std kernel 5.19 of 22.10 following readme here, and it works fine. Probably I should do it this way from the beginning :).

morrownr, 2023-01-20:
Hi @wowa-2017 

> Probably I should do it this way from the beginning :).

Well, yeah, reading the docs can help.

Hey, while I have you. Myself and few others are working on a new version of this driver and it is almost finished. We could use another tester. Are you interested?

Nick

wowa-2017, 2023-01-20:
> > Probably I should do it this way from the beginning :).
> 
> Well, yeah, reading the docs can help.

The matter was I read on some of your pages that it's already included in 6.2 and thought it would be better to have it built-in :).

> Hey, while I have you. Myself and few others are working on a new version of this driver and it is almost finished. We could use another tester. Are you interested?

Sure, please add me to that repo. But it's about out-of-kernel driver? Because it would be interesting as well to understand why built-in version didn't see it.

Vladimir