morrownr/8821cu-20210118 #89: Slow connection and unpredictable behavior with a "Dual Band + Bluetooth USB adapter".

tuqueque, 2023-01-26:
Hello. Yesterday I bought an USB Wifi+Bluetooth adapter for my old desktop computer hoping it could work in a "plug and play" manner like the old USD WiFi adapter I'm replacing... I didn't know I was going to spend 8 hours troubleshooting my system trying to make it work.

I did sort of made it work. At least the WiFi part of it. But is very slow (approximately 6.4 Mbit/s, instead of the at least 50 Mbit/s my Internet connection allows)... Also, in my tests so far, sometimes when I restart my computer I need to unplug and plug the USB adapter back in for the system to recognize it again.

The Bluetooth part is worse. It appears to work, it finds my Bluetooth earphones, but 9 out of 10 times it fails to pair with them and when it does, the connection fails after 5 seconds.

At least I can connect to the Internet, which is progress...

Anyway... I'm not a developer and I would consider myself a mid-level Linux user at best, but if there's anything I can do as an user to help troubleshoot problems with these drivers, let me know.

My system: Manjaro XFCE
Kernels I've tried: 6.1.7-1 and 6.2rc4-2

lsusb output:
`Bus 002 Device 010: ID 0bda:c820 Realtek Semiconductor Corp. 802.11ac NIC`

inxi -Fxxxz && rfkill list && mokutil --sb-state ourput:
```
System:
  Kernel: 6.2.0-2-MANJARO arch: x86_64 bits: 64 compiler: gcc v: 12.2.1
    Desktop: Xfce v: 4.18.0 tk: Gtk v: 3.24.35 info: xfce4-panel wm: xfwm
    v: 4.18.0 vt: 7 dm: LightDM v: 1.32.0 Distro: Manjaro Linux
    base: Arch Linux
Machine:
  Type: Desktop Mobo: ASRock model: G41M-VS3. serial: <superuser required>
    BIOS: American Megatrends v: P1.90 date: 10/01/2013
CPU:
  Info: quad core model: Intel Core2 Quad Q9550 bits: 64 type: MCP
    smt: <unsupported> arch: Penryn rev: A cache: L1: 256 KiB L2: 12 MiB
  Speed (MHz): avg: 2743 high: 2826 min/max: N/A cores: 1: 2660 2: 2660
    3: 2826 4: 2826 bogomips: 22612
  Flags: ht lm nx pae sse sse2 sse3 sse4_1 ssse3 vmx

(...)

Network:
  Message: No PCI device data found.
  IF-ID-1: wlp0s29f7u7i2 state: up mac: <filter>
  IF-ID-2: ztwdjckutl state: unknown mac: <filter>
Bluetooth:
  Device-1: Realtek 802.11ac NIC type: USB driver: btusb,rtl8821cu
    bus-ID: 2-7:10 chip-ID: 0bda:c820 class-ID: e001 serial: <filter>
  Report: rfkill ID: hci0 rfk-id: 2 state: up address: see --recommends
```
Greets and thank you for your hard work!

morrownr, 2023-01-26:
Hi @tuqueque 

I'll start off with some advice based on helping users for a few years.

This driver is stable and fast. Generally the only people that have problems are the ones with a multi-function adapter. That is, the adapter supports wifi and bt. It seems there may be firmware issues with some multi-function adapters. I have seen some users sort things out by using a specific driver option but really, if you can get your money, I recommend that you do.

What you can do is go over to the Main Menu:

https://github.com/morrownr/USB-WiFi

Read menu items 1 and 2... and as many more as your wish and if you have questions, post in `issues` in USB-WiFi.

With that said, we can fiddle around and try to get the adapter working better if you want.

Nick

tuqueque, 2023-01-28:
Hey, thank you for the feedback. I read the items you suggested. Those could be useful in the future...

I cannot return the USB adapter, so I want to make it work better if possible. So any help would be appreciated!

One odd thing I also notice with my WiFi USB adapter is that while download speed is limited, upload speed is faster... 2 or 3 times faster than download speeds (my Internet service allows symmetrical download/upload speeds).

BTW, the version of the driver I'm using is the one from the AUR: https://aur.archlinux.org/packages/rtl8821cu-morrownr-dkms-git ... It seems to be well maintained, so I think building it myself from sources wouldn't change things... Am I correct?

I also read in the following in the main page here:
"Note: As of Linux kernel 6.2, an in-kernel driver for the chipsets supported by this driver has been included in the Linux kernel. The installation and removal scripts for the driver in this repo automatically deactivate the in-kernel driver on installation and reactivate the in-kernel driver on removal. No special action needs to be taken by users."

From what I can understand, version 6.2 of the Linux kernel has drivers included very much like these ones, so theoretically, if I use kernel 6.2, I can uninstall these drivers and the kernel should take care of things automatically... correct?... Well, I tried that when I was trying to make things work a couple of days ago and it didn't work, so I had to install these drivers again.

As I've said before, I consider myself a mid-level Linux user, so I don't know if I'm not understanding some things or if I could do some things differently to improve the connection speed of my device. Again, any help would be welcome.

Finally, I know answering my following question would be just speculation, but... considering my WiFi USB device seems to be fairly new (in the package it came in, it says is compatible with Windows 11 among other things, so it must be a fairly new product)... Would it be too optimistic to think that compatibility could improve over the coming months as these drivers evolve?

morrownr, 2023-02-02:
Hi again @tuqueque 

This repo will be going down soon. The new driver is now available:

https://github.com/morrownr/8821cu-20210916

Please remove this driver and go to the new repo. We can pick up the problem over there.