morrownr/8821cu-20210118 #90: Kernel error on 6.1.8 with Realtek RTL8822CU in Sparklan WPET-239ACN

bchemel, 2023-01-27:
Trying to get [this card](https://www.sparklan.com/product/wpet-239acnbt-series-half-minipcie-wifibt-combo-rtl8822cu/) - a mini PCIe adapter with a Realtek RTL8822CU - running on one of our Arch-based appliances. The current `HEAD` ([a0c1897](https://github.com/morrownr/8821cu-20210118/commit/a0c18978d1c9ab89f96083354f2c55cf15d483f7)) compiles and installs with no problems via `sudo ./install-driver.sh`, but after reboot I get a null pointer dereference in dmesg.

I don't see any obvious matches in the current Issues list, so thought I'd open a new one in the hopes that someone has seen this before. Any help appreciated!

**System info:**
```
uname -a
Linux verve-gateway-22120026 6.1.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Tue, 24 Jan 2023 21:07:04 +0000 x86_64 GNU/Linux
```

**Dmesg excerpt**
```
[    4.057594] RTW: module init start
[    4.057597] RTW: rtl8821cu v5.12.0-16-g5229e7f95.20210118_COEX20200730-5151
[    4.057599] RTW: build time: Jan 27 2023 17:00:50
[    4.057600] RTW: rtl8821cu BT-Coex version = COEX20200730-5151
[    4.068269] e1000e 0000:00:1f.6 eth0: (PCI Express:2.5GT/s:Width x1) ac:40:ea:02:95:e1
[    4.068275] e1000e 0000:00:1f.6 eth0: Intel(R) PRO/1000 Network Connection
[    4.068386] e1000e 0000:00:1f.6 eth0: MAC: 13, PHY: 12, PBA No: FFFFFF-0FF
[    4.068534] RTW: [HALMAC]55772M
               HALMAC_MAJOR_VER = 1
               HALMAC_PROTOTYPE_VER = 6
               HALMAC_MINOR_VER = 6
               HALMAC_PATCH_VER = 22
[    4.068743] BUG: kernel NULL pointer dereference, address: 0000000000000000
[    4.068817] #PF: supervisor instruction fetch in kernel mode
[    4.068870] #PF: error_code(0x0010) - not-present page
[    4.068917] PGD 0 P4D 0
[    4.068946] Oops: 0010 [#1] PREEMPT SMP NOPTI
[    4.068988] CPU: 2 PID: 247 Comm: systemd-udevd Tainted: G           OE      6.1.8-arch1-1 #1 d43e9bdcfe4ce09800bbe1da2a184cb293bc3e00
[    4.069097] Hardware name: Default string DWL01/DWL01, BIOS 5.13 07/31/2020
[    4.069160] RIP: 0010:0x0
[    4.069197] Code: Unable to access opcode bytes at 0xffffffffffffffd6.
[    4.069255] RSP: 0018:ffffb413805b3b18 EFLAGS: 00010246
[    4.069305] RAX: 0000000000000000 RBX: ffff911b04ef0000 RCX: ffffffffc11f4173
[    4.069369] RDX: ffffffffc11f4173 RSI: 0000000000000000 RDI: ffff911b077acc00
[    4.069434] RBP: ffffffffc12b3cc0 R08: ffffffffc11b2530 R09: ffff911b035e3000
[    4.069497] R10: 0000000000000000 R11: 0000000000000000 R12: ffff911b05f61c00
[    4.069560] R13: ffff911b05f4c688 R14: ffffffffc127e860 R15: ffff911b05f61c00
[    4.069612] FS:  00007fa497c4e080(0000) GS:ffff911e5bd00000(0000) knlGS:0000000000000000
[    4.069674] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    4.069715] CR2: ffffffffffffffd6 CR3: 000000010027e006 CR4: 00000000003706e0
[    4.069764] Call Trace:
[    4.069790]  <TASK>
[    4.069814]  rtw_halmac_init_adapter+0xf9/0x31b [8821cu 09090ead228ce69bda4b6434bfc65eb7c298b44a]
[    4.070125]  rtl8821cu_halmac_init_adapter+0x2f/0x34 [8821cu 09090ead228ce69bda4b6434bfc65eb7c298b44a]
[    4.070408]  rtl8821cu_set_hal_ops+0x12/0x152 [8821cu 09090ead228ce69bda4b6434bfc65eb7c298b44a]
[    4.070685]  rtw_set_hal_ops+0x39/0x4e [8821cu 09090ead228ce69bda4b6434bfc65eb7c298b44a]
[    4.070932]  rtw_usb_primary_adapter_init+0xc4/0x2d1 [8821cu 09090ead228ce69bda4b6434bfc65eb7c298b44a]
[    4.071186]  rtw_drv_init+0x4c6/0x695 [8821cu 09090ead228ce69bda4b6434bfc65eb7c298b44a]
[    4.071424]  usb_probe_interface+0xe3/0x2b0
[    4.071469]  really_probe+0xdb/0x380
[    4.071505]  ? pm_runtime_barrier+0x54/0x90
[    4.071542]  __driver_probe_device+0x78/0x170
[    4.071579]  driver_probe_device+0x1f/0x90
[    4.071611]  __driver_attach+0xd2/0x1c0
[    4.071643]  ? __device_attach_driver+0x110/0x110
[    4.071679]  bus_for_each_dev+0x88/0xd0
[    4.071711]  bus_add_driver+0x1b2/0x200
[    4.071744]  driver_register+0x8d/0xe0
[    4.071779]  usb_register_driver+0x88/0x130
[    4.071814]  ? 0xffffffffc0f06000
[    4.071849]  rtw_drv_entry+0xf1/0x1000 [8821cu 09090ead228ce69bda4b6434bfc65eb7c298b44a]
[    4.072063]  ? 0xffffffffc0f06000
[    4.072095]  do_one_initcall+0x5a/0x220
[    4.072132]  do_init_module+0x4a/0x1e0
[    4.072157]  __do_sys_init_module+0x17f/0x1b0
[    4.073904]  do_syscall_64+0x5c/0x90
[    4.075581]  ? syscall_exit_to_user_mode+0x1b/0x40
[    4.077275]  ? do_syscall_64+0x6b/0x90
[    4.078897]  ? exc_page_fault+0x74/0x170
[    4.080624]  entry_SYSCALL_64_after_hwframe+0x63/0xcd
[    4.082182] RIP: 0033:0x7fa497f2216e
[    4.083716] Code: 48 8b 0d 1d ec 0c 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 ca b8 af 00 00 00 0f 05 <48> 3d
01 f0 ff ff 73 01 c3 48 8b 0d ea eb 0c 00 f7 d8 64 89 01 48
[    4.087395] RSP: 002b:00007ffef4a77248 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
[    4.089227] RAX: ffffffffffffffda RBX: 000055f7e899ecd0 RCX: 00007fa497f2216e
[    4.091080] RDX: 000055f7e893adf0 RSI: 000000000063ce65 RDI: 00007fa494efa010
[    4.092959] RBP: 000055f7e893adf0 R08: 000000000065d000 R09: 9e3779b185ebca87
[    4.094493] R10: 000000000005df61 R11: 0000000000000246 R12: 0000000000020000
[    4.096333] R13: 000055f7e899d140 R14: 000055f7e899ecd0 R15: 000055f7e89a1e40
[    4.098185]  </TASK>
[    4.099708] Modules linked in: intel_cstate pcc_cpufreq(-) fjes(-) intel_rapl_msr(+) btrtl snd_timer i2c_i801 btbcm intel_uncore wmi_bmof pcspkr rndis_h
ost 8821cu(OE+) e1000e i2c_smbus snd btintel cdc_ether processor_thermal_device_pci_legacy mei_me soundcore btmtk processor_thermal_device processor_therma
l_rfim qrtr igb usbnet processor_thermal_mbox dca processor_thermal_rapl mei bluetooth mii cfg80211 intel_rapl_common ecdh_generic int340x_thermal_zone rfk
ill intel_soc_dts_iosf intel_pch_thermal vfat fat acpi_pad acpi_tad mac_hid fuse bpf_preload ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 i915
 drm_buddy intel_gtt drm_display_helper xhci_pci xhci_pci_renesas crc32c_intel cec video ttm wmi
[    4.105811] CR2: 0000000000000000
[    4.107123] ---[ end trace 0000000000000000 ]---
```

morrownr, 2023-01-27:
@bchemel 

I just adding you to a testing repo. We have new code for this driver and have been getting it ready for about 2.5 months now. The plan is to open the repo to the public on 2023-02-02, which is next week. We would then soon close down this repo so let's let this repo do its thing and concentrate on the new driver.

You should get an email that tells you what to do. Come on and test the driver and give us a report. Thanks.

Nick

bchemel, 2023-01-27:
Awesome! Thank you!

morrownr, 2023-01-28:
Done